OpenwaterHealth/SlicerOpenLIFU #4: Add the ability to load a session

ebrahimebrahim, 2024-06-21:
It should be possible to read a session json file and have that load into Slicer an MRI volume, a transducer geometry, and a target location.

- Study the [openlifu](https://github.com/OpenwaterHealth/OpenLIFU-python) notebooks to see an example of working with session data and processing it.
- See the [user manual of the matlab tool](https://github.com/OpenwaterHealth/opw_neuromod_sw/wiki/1.-User-Manual) to get a sense for workflow and expected UI

ebrahimebrahim, 2024-06-21:
@peterhollender Please check this issue description and edit freely :)

We talked about loading a session from a file. There is also the concept of a session _browser_, where you can see a list of patients, click on a patient, and then see a list of sessions to choose from. I didn't include that here -- we can make that a separate issue. For now this issue is just about loading a single session. I wasn't totally clear on this: do we want a session browser yet (based on some folder hierarchy) or do we want to hold off until work gets under way for the data management platform?

peterhollender, 2024-06-23:
We will want a session browser, but we could have a placeholder file browser window that pops up to have the user select the session json file, and then just interprets the subject_id and session_id from that file's location and contents, before implementing the session browser. That said, the `Database` class does have methods to get all of the subject IDs and get all the session IDs for a subject, so the browser may not be too complicated and modifying the Database class to access files differently won't change its use in this context

ebrahimebrahim, 2024-08-15:
closed by #50

closed 2024-08-15 by ebrahimebrahim