OpenwaterHealth/SlicerOpenLIFU #60: Implement sonication planning module controls

ebrahimebrahim, 2024-08-15:
Depends on #56 and #57.
Depends on #58 as well but could still get started without it.

Implement some of the functionality in OpenLIFUSonicationPlanning:
- populate dropdowns with protocol, transducer, volume, and target
- add a "plan" button that is just a dummy for now

ebrahimebrahim, 2024-08-26:
For volumes, we can show all volumes using a node selector combobox

And later when we run the simution and generate a beam pattern volume, we can attribute-tag it to prevent it from being displayed here.

ebrahimebrahim, 2024-08-26:
Depends a bit on #65 also

sadhana-r, 2024-08-29:
For volumes and fiducials, can just use qMRMLNodeComboBox. The filtering of nodes is only relevant in Guided Mode, in which case the sonication planning module is actually grayed out since the volume and fiducial nodes are automatically selected internally.  See roadmap [slide](https://docs.google.com/presentation/d/1yH7Nh0XqOb4q_BCZdTR4L2R6TGPShxaMDx2XHCmvxSg/edit#slide=id.g2f2d4b7456b_0_351).

PR #71 (2024-09-06): Implement sonication planning controls
Close #60
Close #73
closed 2024-09-06 by sadhana-r