OpenwaterHealth/SlicerOpenLIFU #98: Add complete solution analysis to SlicerOpenLIFU

ebrahimebrahim, 2024-09-13:
Put the rest of the solution analysis aspects into SlicerOpenLIFU, displaying them in the planning module and limiting the ability to approve a Solution when things are beyond certain bounds. (Where are bounds defined?)

Depends on #97 and https://github.com/OpenwaterHealth/OpenLIFU-python/issues/105

arhowe00, 2025-02-17:
Hey @ebrahimebrahim , I'm not sure whether we got concrete info on the bounds of certain things.  I'm guessing we have

0. Physical limitations (i.e. quantum mechanical or Newton's laws)
1. Hardware limitations
2. Firmware restrictions
3. Software restrictions

where our focus is on 3, and we want to hardcode boundaries that include items 0-2, as well as some preferences laid out by peter.  Should we @ him here?

Planning the implementation part, I think we might want a dedicated that just takes in any parameter (we can dict map this or send the field as a kwarg), and it returns a bool if it is in bound (or we can be clever and return the number itself, otherwise throw).  This way we wouldn't have to have a bunch of boundary checks within the code.

ebrahimebrahim, 2025-02-17:
Hey Andrew,

> I think we might want a dedicated that just takes in any parameter (we can dict map this or send the field as a kwarg), and it returns a bool if it is in bound (or we can be clever and return the [...]

🌵Having a framework in place to handle the final "solution deemed unsafe" and "solution deemed safe" outcomes can indeed be done before https://github.com/OpenwaterHealth/OpenLIFU-python/issues/105 is completed

> I'm not sure whether we got concrete info on the bounds of certain things. I'm guessing we have
> 
> 0. Physical limitations (e.g. quantum mechanical laws, Newton's laws, etc.)
> 1. Hardware limitations
> 2. Firmware restrictions (on the device itself, hardcoded by George)
> 3. Software restrictions
> 
> where our focus is on 3, and we want to hardcode boundaries that include items 0-2, as well as some preferences laid out by Peter. Should we @ him here?

You can see the latest `SolutionAnalysis` class here on the branch where Peter is fleshing it out: https://github.com/OpenwaterHealth/OpenLIFU-python/blob/141-solution-analysis-regression/src/openlifu/plan/solution_analysis.py
🌵He said it's almost done, so that's probably the complete `SolutionAnalysis` class right there. Knowing that, you can actually already write the code that populates the solution analysis table in the planning module!

For specific constraints let's bring it up at the next meeting and get some initial thoughts. I added it to our agenda.

---

Given the two 🌵's above, you can get an early jump on this issue if you want to! I think no other issues touch the same code so it can sit as a PR until https://github.com/OpenwaterHealth/OpenLIFU-python/issues/105 is completed
Up to you that's just if you find time to work on it and have no other pressing things

peterhollender, 2025-02-21:
in the [old MATLAB code](https://github.com/OpenwaterHealth/LOFUgen1), I used a stored set of bounds on fields of the solution analysis, like this: 
```
param_spec = {...
        struct(...
            'field', 'sim_method', ...
            'name', 'Sim Method', ...
            'format', '%s', ...
            'scale', @string, ...
            'warn', @(METHOD)eq(METHOD, "water")), ...
        struct(...
            'field', 'voltage', ...
            'name', 'Voltage', ...
            'units', 'V', ...
            'format', '%0.2f', ...
            'error', @(V)V>35), ...
        struct(...
            'field', 'pnp_MPa', ...
            'name', 'PNP', ...
            'units', 'MPa', ...
            'format', '%0.2f'), ...
        struct(...
            'field', 'mi', ...
            'name', 'MI', ...
            'units', '', ...
            'format', '%0.2f', ...
            'error', @(MI)MI>1.9), ...
        struct(...
            'field', 'isppa_Wcm2', ...
            'name', 'Isppa', ...
            'units', 'W/cm2', ...
            'format', '%0.0f', ...
            'error', @(ISPPA)ISPPA>190), ...
        struct(...
            'field', 'ispta_mWcm2', ...
            'name', 'Ispta', ...
            'units', 'mW/cm2', ...
            'format', '%0.0f', ...
            'error', @(ISPTA)ISPTA>720), ...
        struct(...
            'field', 'TIS', ...
            'name', 'TI', ...
            'units', '', ...
            'format', '%0.2f', ...
            'error', @(TI)TI>6.0), ...
        struct(...
            'field', 'TIC', ...
            'name', 'TIC', ...
            'units', '', ...
            'format', '%0.2f', ...
            'error', @(TIC)TIC>3.0), ...
        struct(...
            'field', 'grating_ratios', ...
            'name', 'Grating Lobes', ...
            'units', '%', ...
            'format', '%0.1f', ...
            'scale', @(x)100*max(x), ...
            'warn', @(RATIO)RATIO>50,...
            'error', @(RATIO)RATIO>60)...
        };
```

the `get_param_table` function would generate for each checked parameter something like this: 
```
        params(i) = struct(...
            'name', p.name, ...
            'value', value, ...
            'units', units, ...
            'format', p.format, ...
            'status', status, ...
            'message', msg);
[... 54 lines cut ...]
                    T.Properties.RowNames, ...
                    T(:,'units').Variables, ...
                    'UniformOutput', false);
                ttlfmt = T.Properties.RowNames;
                app.SimResultsUITable.set(...
                    'Data', [vfmt'; ufmt'], ...
                    'RowName', {'Value', 'Units'}, ...
                    'ColumnName', ttlfmt);
                app.SimResultsTextArea.Value = comment;
                nerr = sum(strcmp({param_table.status},'error'));
                nwarn = sum(strcmp({param_table.status},'warning'));
                app.TreatmentPlanMetricsLabel.Text = ['Treatment Plan Metrics'  repmat(error_icon, 1, nerr)  repmat(warning_icon, 1, nwarn)];
                if nerr>0
                    app.TreatButton.Enable = "off";
                else
                    app.TreatButton.Enable = "on";
                end
            end
        end
```

in `opw_neuromod_sw`, this got wrapped into the `ParamConstraint` class, which is supposed to implement something similar:
[ParamConstraint](https://github.com/OpenwaterHealth/opw_neuromod_sw/blob/main/%2Bfus/%2Btreatment/ParameterConstraint.m)

I guess for Solution Analysis that I ought to add it this formatting and error checking...

ebrahimebrahim, 2025-03-04:
This now has a new dependency: https://github.com/OpenwaterHealth/OpenLIFU-python/issues/230