OpenwaterHealth/SlicerOpenLIFU #143: Create SlicerOpenLIFUPhotoscan

sadhana-r, 2024-11-26:
Create an SlicerOpenLIFUPhotoscan class that enables serialization and the loading of photoscans from json files. The class consists of the model node, texture volume and an approval flag that the user can toggle in the UI.

Depends on [(https://github.com/OpenwaterHealth/OpenLIFU-python/issues/177)]

ebrahimebrahim, 2024-12-04:
> that the user can toggle in the UI

Where would this toggle go?
If it's not clear at this time, then I think we can leave this UI aspect out of this issue. Photoscan approval would be part of the more refined transducer tracking workflow that we will pin down more clearly later.

---

This issue can focus on
- creating SlicerOpenLIFUPhotoscan with a static "initialize from openlifu data" method
- having a list of photoscans in the openlifu data module parameter node, and showing the loaded photoscans in the loaded objects view
- having a "load photoscan" button in the openlifu data module for manual loading of photoscans, allowing the loading of an obj file directly or of a json file (similar to how "load volume" works and allows the loading of a volume data file or of a json file)

So this issue can focus only on the "manual" workflow, nothing session-related and no need to touch the transducer tracking module

sadhana-r, 2024-12-04:
> > that the user can toggle in the UI
> 
> Where would this toggle go? If it's not clear at this time, then I think we can leave this UI aspect out of this issue. Photoscan approval would be part of the more refined transducer tracking workflow that we will pin down more clearly later.
> 
I was envisioning the toggle button being a part of the "Photoscan preview" in the transducer tracking UI, that opens up when a "View photoscan" option is clicked. we can discuss this more but the preview window could also open up when the photoscan is loaded (in the manual workflow). But I agree, this issue can focus on the manual worlflow and we can figure out the transducer tracking UI and add in the preview window in a separate issue.

PR #158 (2025-02-10): Create SlicerOpenLIFUPhotoscan and implement manual workflow
Closes #143 
Closes #147 
Closes #160 

This PR should be merged after PR #155. 

- [x] Create SlicerOpenLIFUPhotoscan
- [x] Add load photoscan button for user to select model or json file
- [x] Load openlifu photoscan using either json or model/texture filepaths and create SlicerOpenLIFU Photoscan object
- [x] Have a list of photoscans in the openlifu data module parameter node, and showing the loaded photoscans in the loaded objects view
- [x] Populate the list of photoscans in the transducer tracking module using these SlicerOpenL [... cut]
closed 2025-02-10 by ebrahimebrahim