OpenwaterHealth/SlicerOpenLIFU #149: Implement end-to-end transducer tracking

ebrahimebrahim, 2024-12-04:


ebrahimebrahim, 2025-01-29:
New way of approaching this: Implement the manual transducer tracking protocol _first_.

Notes here: https://docs.google.com/document/d/1E9hJlHqPt-KLCHQNYtSKhpBEe1FLfrtNmtYeeNP1Azo/edit?tab=t.0#heading=h.k6n9lcitkf0o

We will gradually automate the 3 major time consuming manual steps:
- Putting facial landmarks on the photoscan
- Putting landmarks on the mri skin seg
- Putting transducer landmarks on the photoscan

ebrahimebrahim, 2025-02-04:
Remember to also remove the temporary "select skin segmentation model" dropdown and replace it by the skin segmentation introduced in https://github.com/OpenwaterHealth/OpenLIFU-python/pull/190

ebrahimebrahim, 2025-02-11:
@sadhana-r  Peter has provided a list of which aspects of "manual transducer tracking" they hope to have by April 1:

> 1. Extraction of the skin surface from the MRI
> 2. Guided marking of the Eyes and Nasion
> 3. Guided marking of the Eyes and Nasion on the Photoscan
> 4. Fiducial and Surface Registration (similarity) of the Photoscan to the MRI
> 5. Surface registration of the transducer to the Photoscan, using the virtual fit location to initiate the search.

And this is the doc where we have the latest written summary of the manual workflow: https://docs.google.com/document/d/1rWI5ZUjMTd-zCz1qryXEaxuapIDLFfWw9b7KF4eu-tA/edit?usp=drive_link

ebrahimebrahim, 2025-02-27:
To add some comments from today's meeting, for our first end-to-end transducer tracking:
- Fiducial based registration only is sufficient for the photoscan <--> MRI registration. The ICP step can be a placeholder that returns the identity transform for example
- For the transducer <--> photoscan registration, it is okay for now to have a placeholder in which the user is just given thhe transform handles to manually move the transducer into place.

sadhana-r, 2025-03-11:
This issue has been broken down into the following, smaller issues:

- #199 
- #200 
- #201
- #202
- #203
- #196

closed 2025-04-09 by sadhana-r