OpenwaterHealth/SlicerOpenLIFU #164: Map Protocol Configuration UI to Logic Functions

arhowe00, 2025-01-28:
## Description

Enhance the `OpenLIFUProtocolConfig` module by implementing the following:

- Map widget buttons to new logic functions for protocol management.
- Add logic functions to get/set protocol values.
- Connect UI widgets to logic functions, including:
  - Populate a dropdown menu in the protocol configuration module with currently loaded protocols (from the `OpenLIFUData` module’s parameter node), including a "create new" option.
  - Implement save protocol functionality with overwrite dialog support.
  - Add delete protocol functionality.
  - Dynamically adjust widget availability based on values of certain widgets (e.g., `FocalPattern` determines available focal pattern parameters).

arhowe00, 2025-01-31:
When a user is editing a selected protocol in the config, there are two possible behaviors:
1. The selected protocol is edited and, in real time, the protocol in memory changes to reflect what is configured. This would occur even if the user does not hit "Save Protocol."  Then, if the user hits "Save Protocol," the selected protocol will be written to the database.
2. The selected protocol is edited, but nothing changes in memory, nor in the database, until "Save Protocol" is hit.  This means that the configuration would be adjusted until desired, and the protocol in memory only changes when "Save Protocol" is hit, which will invoke two functionalities: a) modifying the protocol loaded in memory and b) writing that protocol to the database.
Which of these behaviors do you think I should implement @ebrahimebrahim ?

arhowe00, 2025-01-31:
Similar to above, what someone hits "Create new Protocol," should anything be written to the DB?  What about protocols loaded in memory?

ebrahimebrahim, 2025-01-31:
> When a user is editing a selected protocol in the config, there are two possible behaviors:
>
> 1. The selected protocol is edited and, in real time, the protocol in memory changes to reflect what is [...]

There's
- (A) protocol on disk (in openlifu database)
- (B) protocol in memory (serialized into the data module's parameter node)
- (C) protocol in qt model (the values in the widgets in the protocol config module)

What happens when (C) changes?
Updating all the way to (A) every time (C) changes is bad. Too many writes and also we would commit half-written protocols into the database -- not good.
So then do we update (B) only when (C) changes? That's also not good, because a discrepancy between (B) and (A) sounds like a recipe for lots of confusion. Not a friendly experience.
Then there's only one way: Nothing updates when (C) changes.

When the user hits "save", then the chain of updates happens (C) -> (B) -> (A)
So your option (2).

But there is a problem: When the user has changed (C) and hasn't yet hit _save_, the displayed state of (C) no longer matches the underlying (B) that would actually end up getting used. This is dangerous.

So we need a way to make it clear to the user that they currently have _unsaved changes_ in the protocol configuration module. It should be super super clear somehow. And there should probably be some way to discard current changes.

ebrahimebrahim, 2025-01-31:
> Similar to above, when someone hits "Create new Protocol," should anything be written to the DB?

I think it's easiest to treat it the same way. "Create new protocol" puts (C) into a blank state for you, even allowing you edit ID. (Normally you should not be able to edit ID since that would _be_ creating a new protocol, right?)

Only when you hit "save" does the update (C)->(B)->(A) happen

> What about protocols loaded in memory?

What do you mean? What about them?

arhowe00, 2025-02-03:
I've mostly addressed the problem of choosing when to write something in memory and in the database, using a simple state machine which you can see in the ongoing commits.  I'm pretty much through implementing all the buttons, but I'm now addressing the issue of creating new protocols. I think it makes intuitive sense for newly created protocols to be selectable but unsaved to both memory and the database, and for the user to have as many unsaved protocols in progress that they want as long as the application is open.  I have an idea of how I would address this, but I want to know whether implementing this at all is the way I should go (I could just make it so that "Create New Protocol" doesn't do anything to the selector until it is saved in the database and loaded).  If I go the route that I think makes sense, it would involve storing a list of "ephemeral" protocols in the logic memory, but not in the application memory or database. They can be selected from the drop down and maybe are appended with (UNSAVED).

To think about this, it might lead to making every protocol with desired edits ephemeral and unsaved until the save button is pressed.  That way somebody can edit a protocol, switch to another with unsaved changes, and switch back. The "Saved changes" state machine will be easy to implement with this, but it will require a sort-of caching mechanism for unsaved protocols.

Thoughts?

ebrahimebrahim, 2025-02-03:
The ephemeral protocols idea is one approach that makes sense, but I wonder if it's overcomplicating things without adding too much of value? Is it important to be able to leave a protocol in a half saved state while I go do other things in the application?

Well... maybe you'd want to check other protocols to reference them while you build your new one...

Okay that does sound pretty useful... I think it comes down to how much complexity it's adding.

It sounds like you've thought through how to do it. If you think it's not too too complex given what you already have, then I think it's a good idea to do it. I totally agree with the ideas of appending "(UNSAVED)" to the name and not putting them into the application memory but just in the protocol config module's ephemeral protocols cache.

If you do feel it's adding a lot of complexity then maybe it can be avoided: you can instead just insist that the user finish their edits and save before going off and viewing another protocol. Not as friendly but still not bad.

commit 816fe4f2 (2025-02-04): Remove construction notice from protocol config (#164)

Closes #164

- The initial implementation of the protocol configuration module has
  been completed.
commit d4d2985d (2025-02-05): Remove construction notice from protocol config (#164)

Closes #164

- The initial implementation of the protocol configuration module has
  been completed.
commit e7b2c595 (2025-02-10): Remove construction notice from protocol config (#164)

Closes #164

- The initial implementation of the protocol configuration module has
  been completed.
PR #168 (2025-02-11): Implement barebones protocol config module
Closes #164
closed 2025-02-11 by sadhana-r