OpenwaterHealth/SlicerOpenLIFU #165: Add Additional Protocol Parameter Controls

arhowe00, 2025-01-28:
## Description

Determine which additional protocol parameter controls should be exposed and add them to the UI.

### References

- Depends on #163

ebrahimebrahim, 2025-02-07:
Adding to this issue the following:

- Write units on the protocol control module fields that represent physical quantities
- Put tooltips explaining what the fields are. A good way to do this would be to steal the docstrings from the `Protocol` class itself, [once they exist](https://github.com/OpenwaterHealth/OpenLIFU-python/issues/201).
- Enforce range limits on the numerical protocol fields (e.g. via suitable QValidators)

@peterhollender please inform about reasonable range limits

ebrahimebrahim, 2025-03-05:
Yes I think the idea is to show the units in the GUI.
So for example instead of this saying "Pulse frequency"
https://github.com/OpenwaterHealth/SlicerOpenLIFU/blob/54f857afce5d67b93937255a8b100697b5207fd6/OpenLIFUProtocolConfig/Resources/UI/OpenLIFUProtocolConfig.ui#L119
it would say "Pulse frequency (Hz)", which we know because of the docstrings and/or comments we can find in openlifu:
https://github.com/OpenwaterHealth/OpenLIFU-python/blob/322c75a3ea7b3eeba93aeb4bea7ab092412cf4a6/src/openlifu/bf/pulse.py#L12-L23

ebrahimebrahim, 2025-03-05:
> * Enforce range limits on the numerical protocol fields (e.g. via suitable QValidators)

Yes It needs some input from Peter I think, but once you have everything in place and all the QValidators in there then you can just quickly ask what would be reasonable numbers to use. These are not safety limits, I think they are more just general reasonableness limits.

peterhollender, 2025-03-05:
Hmm... enforcing of "reasonable ranges" here is actually a bit tricky. The requirement is frequency 150-500kHz (easy enough), Mechanical Index 0-1.9 (which would involve checking the center frequency and the target pressure), Pulse length is 0-100ms (easy enough) max duty cycle is 50% (which means checking pulse length and pulse interval in the sequence) and max total duration is 15 minutes (pulse interval * number of pulses * number of pulse trains * pulse train interval)...

ebrahimebrahim, 2025-03-05:
> Hmm... enforcing of "reasonable ranges" here is actually a bit tricky. The requirement is frequency 150-500kHz (easy enough), Mechanical Index 0-1.9 (which would involve checking the center [...]

In that case maybe it's best to actually have a protocol reasonableness validator function within OpenLIFU-python, rather than UI-level validation.

We could also do both. Super simple UI-level validation that just enforces a wide but finite range within each input box individually, and also a complicated overall protocol validation that is only called when you go to write your new protocol to the database.

peterhollender, 2025-03-05:
Yeah, that's a good idea. Since you can write whatever you want in a .json file, we should enforce limits on the OpenLIFU-python side

arhowe00, 2025-03-26:
@ebrahimebrahim @peterhollender :

> In that case maybe it's best to actually have a protocol reasonableness validator function within OpenLIFU-python, rather than UI-level validation.

These validators were implemented in https://github.com/OpenwaterHealth/OpenLIFU-python/pull/250, right? Via `param_constraint.py`? So to accomplish this:

> Enforce range limits on the numerical protocol fields (e.g. via suitable QValidators)

This issue, as well as #98  also depends on https://github.com/OpenwaterHealth/OpenLIFU-python/issues/230?

peterhollender, 2025-03-26:
> [@ebrahimebrahim](https://github.com/ebrahimebrahim) [@peterhollender](https://github.com/peterhollender) :
> 
> > In that case maybe it's best to actually have a protocol reasonableness validator function within OpenLIFU-python, rather than UI-level validation.
> 
> These validators were implemented in [OpenwaterHealth/OpenLIFU-python#250](https://github.com/OpenwaterHealth/OpenLIFU-python/pull/250), right? Via `param_constraint.py`? So to accomplish the following item in this issue [#165](https://github.com/OpenwaterHealth/SlicerOpenLIFU/issues/165) :
> 
> > Enforce range limits on the numerical protocol fields (e.g. via suitable QValidators)
> 
> this issue depends on [OpenwaterHealth/OpenLIFU-python#230](https://github.com/OpenwaterHealth/OpenLIFU-python/issues/230) being merged?

No, not exactly. Parameter Constraints are data that are _part of the protocol_, and set acceptable ranges for solution-specific computed values for that Protocol. These may include safety limits, dosing minima, or indicators of a failed solution planning (such as due to bad positioning of the transducer). The types of validation we're talking about here are on the values that you are allowed/able to use when _creating_ such a protocol. Some of those may partially overlap with Parameter Constraints, but would be fundamentally different - for example, because we are a Low-Intensity Focused Ultrasound system, we should not allow users to _create a protocol_ that specifies a target pressure corresponding to a Mechanical Index (MI) > 1.9, but on the flip side, lets say that we generate a protocol that targets a lower MI = 1.0. We may also set a Parameter Constraint so that if during the solution planning process, a solution is generated where analysis of the simulation results tells us that it will have a MI > 1.0, the software will flag that result and block sonication. 

I have not written the protocol validation scheme yet - do you need me to?

arhowe00, 2025-03-26:
> I have not written the protocol validation scheme yet - do you need me to?

Yes, and I linked PR 230 because you created that class in `param_constraint.py` that seems to generically solve this problem. Furthermore, you changed the protocol class to include one, so I thought you were hinting at doing it on that branch.

peterhollender, 2025-03-26:
> Yes, and I linked PR 230 because you created that class in `param_constraint.py` that seems to generically solve this problem. Furthermore, you changed the protocol class to include one, so I [...]

Re-read my previous comment. Parameter Constraints are for solution validation, not protocol validation.

peterhollender, 2025-03-26:
We could (and probably should) add configuration of the ParameterConstraint dictionary to the protocol builder. Although that will also require even further meta-constraints on what the constraints can be set to...

peterhollender, 2025-03-26:
I suppose the ParameterConstraint class could be extended in use to be applied for meta validation of protocol inputs (including meta validation of the values used in the ParameterConstraint dictionary). But at the same time, some of the protocol value validation may involve more complicated math than what is encoded into a simple "less than", "greater than", "in between" comparison. JSON serialization/deserialization of more complicated calculations could get messy. 

We may want to also like... be able to pop out the JSON in a text editor? Or try to instantiate the protocol when you save and return exception text?

ebrahimebrahim, 2025-03-27:
> Pardon my missing the point, I thought we could repurpose the class in there to also decide what you can/can't create. So in the simpler case, you would just provide a set of constants that restrict what values can be accepted in the protocol creator?

I think Peter earlier suggested putting the validation into `Protocol`'s __post_init__ method. We could do that,  or we can just make a `Protocol.validate()` method. It will just be a function that returns bool telling you if the protocol is valid (valid here meaning just generally reasonable/possible, more of a sanity check than anything about safety). That doesn't need to be done yet in the present issue. @arhowe00 Here in this issue you can focus on adding the additional controls, with tooltips and units. I updated my comment way above to remove the need to do the extensive protocol validation here so we can put that off to later

arhowe00, 2025-03-27:
Note for later: Some constraints on form inputs are interdependent, but thrown exceptions properly notify of the issue:

[image]

closed 2025-04-04 by arhowe00