OpenwaterHealth/SlicerOpenLIFU #172: Create login module

ebrahimebrahim, 2025-02-07:
# Description

Add a Login module.  The login module has a `Username/Password` field that can be filled out; if a matching `User` is found in `get_all_users()`, the application transfers to the Home module with the `user_id` and `roles` set as parameters in the Home module.  Note: If there is no `User` with the `root` role in `get_all_users()`, the user cannot type in a `Username/Password`, and the role is set to `root`.    
 
## How other modules would ask about the login state

There should also be a convenience function in OpenLIFULib somewhere that all other modules can use to ask the login module what are the roles of the currently logged in user. Here is how it can work:

- If user account mode is off, then this would return “root” as the role. If user acct mode is on then we follow the following logic:  
  - If no one is logged in, then the roles returned by this function would be either
    - root, if there is no User with the root role in `get_all_users()`, or
    - some very low-permissions role that we might call  ‘anonymous’, if there exists a root user already.

Then other modules won't have to mess too much with the internals of the login module -- they can get all the info they need out of this convenience function.

## How other modules would be notified when the login state is changed

Not sure what's the best way yet, but if the login state is in the login module's parameter node then other modules can observe that node for changes. Or we can create our own little observer thing. Or maybe ask Sam about this. In any case this problem should be addressed here in this issue.

## Persistence

Ensure reasonable behavior when a slicer scene is saved and loaded. Whether user account mode is on should be preserved, but whether a user is logged in should not – you should have to re-log-in after loading.

# Dependencies

Depends on https://github.com/OpenwaterHealth/OpenLIFU-python/issues/203

arhowe00, 2025-02-11:
1. For now, I'll change `root` to `admin` since `admin` will have all GUI features. Let me know if we should still add `root` as a role.
2. We can add observers on the Login module parameter node, but this may not end up being necessary.

I spoke with @jcfr, and he gave some very helpful advice for handling the enforcement part in #175 , this continues off our discussion about keeping things centralized in the Login module, one thing I will be adding is dependencies to the module which will be important to accessing UI elements of other widgets from the OpenLIFULogin module.

ebrahimebrahim, 2025-02-12:
> 1. For now, I'll change `root` to `admin` since `admin` will have all GUI features. Let me know if we should still add `root` as a role.

Makes sense!

> one thing I will be adding is dependencies to the module

What does this mean?

PR #182 (2025-03-01): Create simple login module
Closes #172 

Some decisions are still to be made, such as whether "Login" should be a button, or whether the fields should be directly in the widget.

If you want to test logging in, the credentials are given thus (`username/password`):
```
complex_user/complex_user_hash
example_admin/example_admin_hash
example_operator/example_operator_hash
example_restricted/example_restricted_hash
```

Note that you will have to update the database to the version in OpenLIFU-Python [PR#215](https://github.com/OpenwaterHealth/OpenLIFU-python/pull/215), as passwords are checked aga [... cut]
closed 2025-03-01 by ebrahimebrahim