OpenwaterHealth/SlicerOpenLIFU #173: Improve login module functionality

ebrahimebrahim, 2025-02-07:
Add a `Reset Password(s)` button and a `Create New User` button to the Login module that do the obvious things

Depends on #172

arhowe00, 2025-03-07:
Okay, instead of adding a `Reset Password(s)` button, I'm going to create a `Manage Account(s)` button that lets passwords be reset, accounts be deleted, etc.  Will be useful if roles are added dynamically and some people need to be assigned/reassigned by admins. Also, since there isn't a "root" role, admins will be able to delete admin accounts too.

arhowe00, 2025-03-07:
Note: Users cannot request password resets because if a user cannot login, we have no way of identifying the user (there is no "e-mail" associated). Therefore, only admins can reset passwords on accounts, arbitrarily.

PR #198 (2025-03-11): Improve login module functionality
Closes #173 

## Summary

This PR adds the critical components needed to interact with the Users' table, such as creating new users, changing passwords, and deleting user accounts. It should be enough for the full app, but some things might be nice to add:

- A way of adding dynamic roles to users (not just operator/admin)
- We'll have to explore how we want admins to interact with dynamic roles (such as ability to view certain protocols). We can also make users editable in a dialog similar to the create new account dialog, but I didn't make that for now beca [... cut]
closed 2025-03-11 by arhowe00