OpenwaterHealth/SlicerOpenLIFU #174: Add user accounts mode toggle

ebrahimebrahim, 2025-02-07:
Add togglable “user account mode”  (maybe there is a better name for it) to SlicerOpenLIFU. The toggle for it can be a button in the login module. Only if user account mode is turned on will the individual OpenLIFU modules later be made to observe the login state and adjust their widget enabledness according to it. This is very similar to [https://github.com/OpenwaterHealth/SlicerOpenLIFU/issues/126](https://github.com/OpenwaterHealth/SlicerOpenLIFU/issues/126) and we should follow the same pattern as that PR here. By default user account mode should be off.

arhowe00, 2025-02-10:
I made an empty login module with the button, and it works (True and False being returned for the respective toggle button settings)

[image]

I do want to make it so that all the modules in this commit actually observe the login module parameter node, do you think this should happen unconditionally of whether we have user account mode on?  I think this makes sense because there isn't a way we can re-initialize connections, and user account mode may be forced at a later time when the app is open.

To start thinking about the logic of en/disabling permissions, I think one thing that makes sense is to actually add entire boxes (i.e. QGroupBox(es) or something) that wrap all the widges and buttons around the permission groups we wish to enforce. I think this makes sense because it decouples enabling and disabling based on user permissions from other reasons you might have in the widget.  If we start playing around directly with buttons, we could end up in an odd situation where they are being re-enabled by the widget.  What do you think about this?  For instance, if we wish a `restricted_user` to have access  to `load protocol` but not `edit protocol`, and then to enable it for `technician`, we could have a permission box (i.e. some <layout> XML tag) around `edit protocol` that is explicitly enabled and disabled based on the user roles that are known to the widget.  Any thoughts on this?

ebrahimebrahim, 2025-02-10:
> To start thinking about the logic of en/disabling permissions, I think one thing that makes sense is to actually add entire boxes (i.e. QGroupBox(es) or something) that wrap all the widges and buttons around the permission groups we wish to enforce. I think this makes sense because it decouples enabling and disabling based on user permissions from other reasons you might have in the widget. If we start playing around directly with buttons, we could end up in an odd situation where they are being re-enabled by the widget. What do you think about this? For instance, if we wish a `restricted_user` to have access to `load protocol` but not `edit protocol`, and then to enable it for `technician`, we could have a permission box (i.e. some XML tag) around `edit protocol` that is explicitly enabled and disabled based on the user roles that are known to the widget. Any thoughts on this [@ebrahimebrahim](https://github.com/ebrahimebrahim) ?

As just discussed, great idea!!
Notes from discussion with everyone:
- set a permissions QWidget (or something like that, probably QWidget) in the .ui files as parent of widgets that we want to disable based on user permission
- the permissions widget should not have a visual effect but just act as a helpful logical thing
- give the permissions widget a dynamic property that says that it's a permissions widget and that says what are the user roles under which it should be enabled
- in the code when we get to #175 , enforce permissions in each module by iterating through all widgets in all openlifu modules and picking out the ones that are tagged via dynamic property as permissions widgets. this would happen every time the login state changes.

ebrahimebrahim, 2025-02-10:
> I do want to make it so that all the modules in this commit actually observe the login module parameter node, do you think this should happen unconditionally of whether we have user account mode on? [...]

I guess you could do it here. What you mean is that you want to add the observers and create a placeholder function like "onLoginParameterNodeChanged" to each module that does nothing for now, correct? That would be fine.

But also there's the idea you mentioned today of putting all the permission enforcement code inside the login module itself. So that would remove the need for all these observers. Only the login module would observe its own parameter node, and then the login module would reach out and manipulate the ui of other modules.

So IDK maybe you could hold off till #175 and decide at that point who has to observe what

commit 5b58ffb1 (2025-02-10): Add empty login module (#174)

Closes #174

- Add a single button to the module for toggling user account mode, as
  well as some utilities for getting the user account mode state from
  the parameter node of the login module.
commit 025c4b88 (2025-02-11): Add empty login module (#174)

Closes #174

- Add a single button to the module for toggling user account mode, as
  well as some utilities for getting the user account mode state from
  the parameter node of the login module.
commit bada8883 (2025-02-13): Add empty login module (#174)

Closes #174

- Add a single button to the module for toggling user account mode, as
  well as some utilities for getting the user account mode state from
  the parameter node of the login module.
PR #177 (2025-02-17): Add empty login module (#174)
Closes #174

- Add a single button to the module for toggling user account mode, as well as some utilities for getting the user account mode state from the parameter node of the login module.