OpenwaterHealth/SlicerOpenLIFU #175: Enforce user permissions

ebrahimebrahim, 2025-02-07:
# Description

Enforce the user permissions by enabling/disabling widgets based on user role. The user role should always be queried via the convenience function introduced in #172.  

Note: The logic here needs to be thought through with care, because widgets can be disabled for other reasons besides login related ones. The devised solution should be something maintainable and not too confusing. The first thing that comes to mind is each widget that needs to care about login state should observe an updateEnabledness function that gets called every time the login state changes. Many widgets already observe a function like this (possibly under a different name) so we’d want to integrate the user permission check into the existing function. Can this be streamlined in some way to reduce code duplication? The names of the “updateEnabledness” functions should be standardized to make them easy to find should we need to ever add a new user role or something like that.  

# Breakdown of this issue by module

This issue will probably best handled by multiple PRs, with each PR focusing on a module or a few modules. Let's use this checklist here to keep track of which modules are done:

- [ ] login
- [ ] data (the SlicerOpenLIFU data module)
- [ ] preplanning
- [ ] transducer tracking
- [ ] planning
- [ ] control
- [ ] protocol config
- [ ] home

# Information on user roles and permissions

This section is information from Peter -- we can request additional details as we get into it.

| Function                       |  Engineering User  | Admin User | System Operator |
|--------------------------------|:------------------:|:----------:|:---------------:|
| Define Array                   |     Python/JSON    |    FALSE   |      FALSE      |
| Run Standalone Simulations     |    Python/Slicer   |    FALSE   |      FALSE      |
| Water Tank Operation           |    Python/Slicer   |    FALSE   |      FALSE      |
| Manage Users                   |    Python/Slicer   |  Admin GUI |      FALSE      |
| Configure and Publish Protocol | Python/Slicer/JSON |  Admin GUI |      FALSE      |
| Create Session / Add Data      | Python/Slicer/JSON |     GUI    |       GUI       |
| Set Target                     | Python/Slicer/JSON |     GUI    |       GUI       |
| Virtual Fitting                |    Python/Slicer   |     GUI    |       GUI       |
| Transducer Tracking            |    Python/Slicer   |     GUI    |       GUI       |
| Create Solution                | Python/Slicer/JSON |     GUI    |       GUI       |
| Run Sonication                 |    Python/Slicer   |     GUI    |       GUI       |

> Engineering Users are not actual application users, but instead interact with the lower level Python and Slicer Components (and in some instances, could use Matlab or anything else to generate the JSON files directly). They would need to have the same OS-level access as an admin user. 
>
> Admin Users have an expanded set of privileges through widgets in the application that are only available to them. These are primarily User and Protocol management, although we may come across other app functions that should be restricted to them.
>
> System Operators are trained personnel given access to key system functionality to plan and run sonications. They cannot change the Protocol(s) that they have access to, nor modify User permissions. For particularly sensitive applications or if the IRB requires it, Operators may be interacting with the app on an OS account that does not have write access to certain data.

# Dependencies

Depends on #172

ebrahimebrahim, 2025-02-10:
See https://github.com/OpenwaterHealth/SlicerOpenLIFU/issues/174#issuecomment-2649289287 for implementation idea when we get to this

arhowe00, 2025-02-11:
Currently thinking more about how we plan this.  So, assuming we implement GUI permissions in logical widgets, and in the login module we uniformly en/disable widgets from the login module based on read user roles (note: we will have to load the widgets in when the login module or app starts up), I think:

1. We should define the user roles within the `User` object (as implemented) as a list of strings (in this case, 3 possible strings)
2. The default state of all widgets is FALSE, except the login.
3. There is a boolean for every single "Function", e.g. `run_sonication=False`.
4. The user role is read from the User object when authenticated and sets corresponding booleans to true. 
5. `not user_account_mode` can just be OR'd with everything.
6. All the widgets with their permission tag (which ideally should be the "Function"?) are set enabled based on (3)

ebrahimebrahim, 2025-02-11:
> * We should define the user roles within the `User` object (as implemented) as a list of strings (in this case, 2 possible strings, with "Engineering User" not pertaining at all to the app)
> * The default state of all widgets is disabled, except the login.

Agreed. It's already a list of strings in https://github.com/OpenwaterHealth/OpenLIFU-python/pull/206

> * There is a boolean for every single "Function", e.g. `run_sonication=False`.
> * ...
> * All the widgets with their permission tag (which ideally should be the "Function"?) are set enabled based on the state of the booleans defined in (3) after the user is processed.

What do you mean? I didn't understand these two points

> 4\. The user role is read from the User object when authenticated and sets corresponding booleans to true (or, the boolean is OR'd with `"admin" in user.roles`).

I think after authentication the login module should set an attribute on its parameter node of type `openlifu.User` that represents the currently logged in user. (Or rather the type might have to be a thin wrapper `SlicerOpenLIFUUser` like we have done for some other things just to enable lazy importing while using parameter nodes)

Whenever a change is detected in the login module parameter node, the `openlifu.User`'s roles are checked, and widgets throughout all openlifu modules can get updated accordingly. No OR'ing needed. Express all the logic in the permission widgets. Plain and simple. Each permission widget lists exactly the roles that would enable it. What do you think?

> 5\. `not user_account_mode` can just be OR'd with everything.

Right, except instead of a literal OR it would be more like 

```py
def updateAllPermissionWidgetEnabledness():
    if not user_account_mode:
        # [go find all the permission widgets and enable them]
        return # return early
    # [go through all permission widgets and conditionally enable them based on the roles they are labeled with]
```

arhowe00, 2025-02-11:
> What do you mean? I didn't understand these two points
By "Function" I am referring to the "Function" column in the table above. I think we should give each widget a logical tag corresponding to the same or similar name that we give to "Function", as we can then toggle based on the boolean.  For example, `run_sonication` should be a property in the widgets that enable that behavior.

> I think after authentication the login module should set an attribute on its parameter node of type `openlifu.User` that represents the currently logged in user. (Or rather the type might have to be [...]

Agree about tracking the user. Yes, we could make everything observe the Login module, _but_, I think it would be possible simply to do everything from the login module. Imagine this:

```
def onUserVerified(self, cur_user): 
    self.manage_users : bool = cur_user.role in ["admin"] or not self.user_account_mode
    self.run_sonication : bool = cur_user.role in ["operator", "admin"] or not self.user_account_mode
    ...
    self.activate_permissions()
...
def activate_permissions():
    some.way.of.accessing.manage_users.widgets.setEnabled(self.manage_users)
    some.way.of.accessing.run_sonication.widgets.setEnabled(self.run_sonication)
```

> > 5. `not user_account_mode` can just be OR'd with everything.
> 
> Right, except instead of a literal OR it would be more like
> 
> def updateAllPermissionWidgetEnabledness():
>     if not user_account_mode:
>         # [go find all the permission widgets and enable them]
>         return # return early
>     # [go through all permission widgets and conditionally enable them based on the roles they are labeled with]

What do you think about the example where everything is in the Login module above, if possible?  I think it has been mentioned that I can directly access the `self.ui` belonging to other widgets from the Home module, as long as we instantiate the widgets.

arhowe00, 2025-02-11:
I see the implementation above relies on a user being used at all, but we can factor parts of that out... I think the general idea I'm proposing is still viable without `onUserVerified` being needed

ebrahimebrahim, 2025-02-11:
> What do you think about the example where everything is in the Login module above, if possible? I think it has been mentioned that I can directly access the `self.ui` belonging to other widgets from the Login module, as long as we instantiate the widgets.

Yes, I liked that idea! I had assumed that's what you meant. It's very nice to keep all the permissions logic in one  place like that.

> By "Function" I am referring to the "Function" column in the table above. I think we should give each widget a logical tag corresponding to the same or similar name that we give to "Function", as we [...]

Oh I see, _that_ Function.

So what I was proposing is that (A) the widget dynamic properties provide the mapping from between widgets and roles, while the permissions expressed in the table are enforced in how we assign roles to widgets.

But you are proposing that (B) the widget dynamic properties provide a mapping between widgets and "Function", while the permissions expressed in the table are enforced in the login module code where you would encode the relation between Functions and roles.

In (A) the widget can have one dynamic property that is simply a string listing any number of roles.
In (B) the widget has several dynamic properties, one for each Function, and each is a bool.

Let me know if I didn't understand (B) correctly

I think I still like (A) more still because it's simple and flexible.
With (A), we express the Functions table above by peppering in the role strings into the ui files appropriately.
So you can kind of "program" that table in by how you put in role strings -- I like that.
The code in the login module would stay fixed if we wanted to adjust permissions -- all that code would do is implement the roles we programmed into the ui files directly.
I like the separation between the implementation of widget en/dis-abling (login module code) and what the actual role specification should be (ui files). It feels like "program logic" versus "configuration data". I like having the configuration data fully live in the ui files and be in a flexible format.

What do you think?

arhowe00, 2025-02-19:
Note for the future: When certain features are disabled, the tooltips are incorrect. We'll have to either remove the tooltips or add some logic for setting the tooltips.  Preferably, we should do this all in the login module. When setting a tooltip for a wrapping QWidget, does that set the tooltip for all its children?  We'll have to figure out a good solution.

PR #185 (2025-03-04): Enforce user permissions
Closes #175

---

## Summary

This PR introduces the general framework for enforcing user roles in the login module by enclosing buttons in permissions widgets. Adding restrictions to buttons and choosing the roles that can interact should be very straightforward.

When testing, try to login and ensure that certain abilities, e.g. editing protocols, can only be done by admins (you can see the full table in [#175](https://github.com/OpenwaterHealth/SlicerOpenLIFU/issues/175)). If it works for protocols, it should work for everything else.

Currently, new users cannot be [... cut]
closed 2025-03-04 by arhowe00