OpenwaterHealth/SlicerOpenLIFU #186: Cannot save sessions that have photoscans

ebrahimebrahim, 2025-02-23:
Trying the current SlicerOpenLIFU main (fafc90c883004a195b1a89c8eeb520fd72c2f52d) and the DVC data at the current OpenLIFU-python main (1dcff787b74a3b301edbab2f52f6d52835bd0198), if I load the mannequin example then I get this error

<details>
<summary>(expand to view error)</summary>

```txt
[Python] Exception raised in onSaveSessionClicked: Unable to write value 'ObservedDict({'scan123': <OpenLIFULib.parameter_node_utils.SlicerOpenLIFUPhotoscanWrapper object at 0x724ee8f1b580>})' of type '<class 'slicer.parameterNodeWrapper.serializers.ObservedDict'>'
Traceback (most recent call last):
  File "/home/ebrahim/openwater/SlicerOpenLIFU/build/lib/Slicer-5.7/qt-scripted-modules/OpenLIFULib/util.py", line 37, in f_with_forwarded_errors
    raise e
  File "/home/ebrahim/openwater/SlicerOpenLIFU/build/lib/Slicer-5.7/qt-scripted-modules/OpenLIFULib/util.py", line 34, in f_with_forwarded_errors
    return f(*args, **kwargs)
  File "/home/ebrahim/openwater/SlicerOpenLIFU/build/lib/Slicer-5.7/qt-scripted-modules/OpenLIFUData.py", line 856, in onSaveSessionClicked
    self.logic.save_session()
  File "/home/ebrahim/openwater/SlicerOpenLIFU/build/lib/Slicer-5.7/qt-scripted-modules/OpenLIFUData.py", line 1238, in save_session
    session_openlifu = session.update_underlying_openlifu_session(targets)
  File "/home/ebrahim/openwater/SlicerOpenLIFU/build/lib/Slicer-5.7/qt-scripted-modules/OpenLIFULib/session.py", line 171, in update_underlying_openlifu_session
    self.target_nodes = targets
  File "/home/ebrahim/slicer-superbuild-8c6219/Slicer-build/bin/Python/slicer/parameterNodeWrapper/parameterPack.py", line 311, in __setattr__
    super().__setattr__(key, value)
  File "/home/ebrahim/slicer-superbuild-8c6219/Slicer-build/bin/Python/slicer/parameterNodeWrapper/parameterPack.py", line 286, in setter
    self._save()
  File "/home/ebrahim/slicer-superbuild-8c6219/Slicer-build/bin/Python/slicer/parameterNodeWrapper/parameterPack.py", line 333, in _save
    serializer.write(parameterNode, name, self)
  File "/home/ebrahim/slicer-superbuild-8c6219/Slicer-build/bin/Python/slicer/parameterNodeWrapper/parameterPack.py", line 431, in write
    parameter.serializer.write(parameterNode, mangledName, parameterValue)
  File "/home/ebrahim/slicer-superbuild-8c6219/Slicer-build/bin/Python/slicer/parameterNodeWrapper/serializers.py", line 161, in write
    self.serializer.write(parameterNode, name, value)
  File "/home/ebrahim/slicer-superbuild-8c6219/Slicer-build/bin/Python/slicer/parameterNodeWrapper/serializers.py", line 938, in write
    index, serializer = self._findBestSerializer(value)
  File "/home/ebrahim/slicer-superbuild-8c6219/Slicer-build/bin/Python/slicer/parameterNodeWrapper/serializers.py", line 934, in _findBestSerializer
    raise TypeError(f"Unable to write value '{value}' of type '{type(value)}'")
TypeError: Unable to write value 'ObservedDict({'scan123': <OpenLIFULib.parameter_node_utils.SlicerOpenLIFUPhotoscanWrapper object at 0x724ee8f1b580>})' of type '<class 'slicer.parameterNodeWrapper.serializers.ObservedDict'>'
```

</details>

I think something might be wrong with `SlicerOpenLIFUPhotoscanWrapper` or the associated serializer? Or maybe something wrong with our DVC example data? Not sure.

I also found that there are slightly different errors related to SlicerOpenLIFUPhotoscanWrapper whenever the data module parameter node's loaded_session gets touched for any reason. For example if you load the mannequin example and then just reassign the loaded session without changing any values, this gives some errors:

```py
session = slicer.modules.OpenLIFUDataWidget.logic.getParameterNode().loaded_session
slicer.modules.OpenLIFUDataWidget.logic.getParameterNode().loaded_session = session
```

But only with the mannequin example, so it seems to be photoscan related.

sadhana-r, 2025-02-24:
Thanks, I'm looking into it. Seems like there is something wrong with the serializer. I don't get any errors when `affiliated_photoscans` is a single `SlicerOpenLIFUPhotoscanWrapper` instead of a List or Dict. I think the seralizer needs to be updated to handle a List or Dict of `SlicerOpenLIFUPhotoscanWrapper`.

ebrahimebrahim, 2025-02-24:
> the seralizer needs to be updated to handle a List or Dict of `SlicerOpenLIFUPhotoscanWrapper`

Hmm but the parameter node wrapper is supposed to create [a serializer](https://github.com/Slicer/Slicer/blob/bcfd05d19e64b0cf87e08bca6f7eb2cedfea0e35/Base/Python/slicer/parameterNodeWrapper/serializers.py#L596) automatically for lists or dicts of things that have a serializer.

PR #187 (2025-02-25): Fix for saving sessions with affiliated photoscans
Closes #186 

Slicer is not able to handle the Optional type annotation for Lists and Dicts of custom serializer classes. As a workaround, `affiliated_photoscans` are no longer an optional attribute of a SlicerOpenLIFUSession, and default to {}.
closed 2025-02-25 by sadhana-r