OpenwaterHealth/SlicerOpenLIFU #257: Virtual fit approvals

ebrahimebrahim, 2025-04-15:
Ensure that virtual fit approval works:
- Toggling approval updates the application state appropriately
- Saving a session saves the approval state
- Loading a session loads the approval state
- Making changes in the scene to the virtual fit transform or its associated target causes the approval to be revoked.

(Most or all of these may already be in place and working, in which case this issue can be closed after a verification of that fact.)

sadhana-r, 2025-04-16:
- [ ] Toggling approval updates the application state appropriately
- [ ] Saving a session saves the approval state
- [ ] Loading a session loads the approval state
- [ ] Making changes in the scene to the virtual fit transform or its associated target causes the approval to be revoked
- [ ] Without an approved virtual fit, the workflow doesn't allow proceeding from the preplanning module (so that would be something that is enforced when guided mode is enabled)

@ebrahimebrahim I tested all of the above uses cases and they work as expected. I'm not sure if this is a situation that would occur in reality, but what are your thoughts on this situation: I compute a VF result for target and approve it. If I modify the location of the target, the approval is revoked but the associated transforms are all still in the scene for a now 'invalid' target. I would assume a user would know to re-run the VF, but technically you could go back and approve the transforms, or save the session but the transforms aren't technically what was computed for the modified target location. I guess my question is whether we need to be providing some indication when transforms are invalidated? or is revoking approval enough?

ebrahimebrahim, 2025-04-17:
> what are your thoughts on this situation: I compute a VF result for target and approve it. If I modify the location of the target, the approval is revoked but the associated transforms are all still [...]

Hmm good question. My sense is that this situation is okay as it currently is, because the _automatic_ virtual fit is just a convenient way to get a transducer placement suggestion. It's not doing any safety checks -- those come later. In the future, I think we will want to allow manual virtual fitting, or manual override of automatic virtual fitting. So the power to approve preplanning really is in the hands of the person doing the preplanning. They can approve whatever transform they want. That's my feeling.

> Without an approved virtual fit, the workflow doesn't allow proceeding from the preplanning module (so that would be something that is enforced when guided mode is enabled)

I think this one probably doesn't hold, from looking at [the workflow logic](https://github.com/OpenwaterHealth/SlicerOpenLIFU/blob/50a6804b582af8ade4118002c9c8ea31d4dd17c1/OpenLIFUPrePlanning/OpenLIFUPrePlanning.py#L500-L511) -- it looks like it lets you proceed if there's a virtual fit result, regardless of whether it's approved

sadhana-r, 2025-04-17:
> Hmm good question. My sense is that this situation is okay as it currently is, because the _automatic_ virtual fit is just a convenient way to get a transducer placement suggestion. It's not doing [...]

I have been thinking about this some more and I think when a target is moved or removed, in addition to virtual fit approval being revoked (if currenly approved), the user should be prompted and asked if they would like to delete the associated virtual fit results or to maybe even confirm deletion. This definitely needs to happen atleast when the target is removed. Because when the user then tries to save the session, an error pops up since the virtual fit results are now associated with a target that is no longer a part of the session. 

I think manually editing the virtual fit is okay, because the user can go back and approve pre-planning if they prefer their manual override of the virtual fit transform. But changing the target could lead to very mismatched transforms?

> I think this one probably doesn't hold, from looking at [the workflow [...]

Oh you're right. My bad - I forgot to specifically check for an approved result. I'll update that

PR #274 (2025-04-25): Update virtual fit approvals workflow
Closes #257 

- In guided mode, you should not be able to proceed to transducer tracking without an approved VF result.
- If a target is removed, approval is revoked (if any) and the associated VF results are cleared.
- If a target is modified (point modified or added/removed), approval is revoked (if any) and the user is asked to confirm deletion of VF results.
- In guided mode, if the virtual fit results are cleared, the workflow status updates accordingly. 

`OnNodeRemoved` is triggered when you unload a session. Since the `loaded_session` gets set to ` [... cut]
closed 2025-04-25 by sadhana-r