OpenwaterHealth/SlicerOpenLIFU #259: Transducer tracking approvals

ebrahimebrahim, 2025-04-15:
Ensure that transducer tracking approval works:
- Toggling approval updates the application state appropriately
- Saving a session saves the approval state
- Loading a session loads the approval state
- Making changes in the scene to the transducer tracking transforms _or the transducer transform_ (because it is what's used in the sonication planner), or the sonication target causes the transducer tracking approvals to be revoked.

ebrahimebrahim, 2025-04-15:
Copying some important questions from #203:

> This includes
> 
> * checking when transducer transforms are modified or removed from the scene.
> * creating a workflow for approval/revoking approval of transducer tracking results
> * When a session has a transducer tracking result approved, if approval of the corresponding photoscan is revoked, the user should be notified. What should happen to the tt result? This is somewhat equivalent to the situation when a target with a virtual fit result is modified/removed.
> * Can a user proceed with transducer tracking when the photoscan is unapproved?

sadhana-r, 2025-04-17:
- In guided mode, cannot proceed to transducer tracking unless the photoscan is approved. This is not enforced when not in guided mode
- Should revoking approval on photoscan revoke transducer tracking approval? Or just block progress  in guided mode and display Warning?
- Currently, unless you approve the transform during the wizard workflow, you cannot go back and approve a previously computed transform. A user should be able to click through wizard and see previous tt results and toggle approval. If a fiducial is moved, tt results are cleared and unapproved (wizard-level). 
- Each step is blocking - cannot proceed to transducer-volume tracking or click 'Finish' without approving the transform.

sadhana-r, 2025-04-25:
> * ~or the sonication target causes the transducer tracking approvals to be revoked.~

After further discussion, transudcer tracking approcal does not need to be revoked upon target modifcation. Target modification already revokes VF approval. If the user has already run transudcer tracking, we will instead include a check before sonication control/planning to ensure that all approvals are in place. So the user will be forced to update the transucer tracking result/approval.

PR #290 (2025-05-01): Update workflow for revoking transducer tracking approval
Closes #259 
Closes #289 

- When a transduer tracking result is computed, the result is copied to the current transducer transform. An observer is added to the current transducer transform so that it revokes tt approval for the associated "matching_transform" when modified.
- When approving a photoscan is a session-based workflow, added a check to enforce that only one photoscan can be approved per session.
- When loading a session, if there is a photoscan (could be approved or unapproved) associated with an approved transducer trackin [... cut]
closed 2025-05-01 by sadhana-r