OpenwaterHealth/SlicerOpenLIFU #262: Add management and enforcement of protocol-specific user permissions

ebrahimebrahim, 2025-04-15:
Once https://github.com/OpenwaterHealth/OpenLIFU-python/issues/239 is done, we will have some scheme for saying which user role or which user has access to which protocols. Whatever scheme we come up with, we should add (1) a method for managing that in the SlicerOpenLIFU user management interface and (2) enforcement of that information based on who is the currently logged in user. So whether it's visibility of protocols that we decide should be enforced, or whether it's the ability to use them to initiate a session, or whether it's both of those things, we should enforce that.

There is also (3): a method for managing this in the protocol configuration module when creating or editing protocols. However that's not part of this issue here as it's to be included in a future milestone.

Depends on https://github.com/OpenwaterHealth/OpenLIFU-python/issues/239

arhowe00, 2025-04-24:
Adding (3) will be trivial (a few lines of code) due to the existing infrastructure. I can knock it all out in one PR unless you prefer using a separate issue.

arhowe00, 2025-04-24:
To implement (2), it may make sense to unload all data when changing users. Since the login comes before loading data, I think it is reasonable to wipe the scene and all loaded data. However, it will make (2) easier because we can just prevent loading of protocols that are not allowed by the user. If we were not to wipe all the data, it would add application state that need special consideration, such as:

1. A protocol loaded by one allowed user cannot be loaded by another, but it was already loaded (reason for wiping protocols)
2. A protocol loaded by one allowed user was used to compute session data, but another user logs in who is not allowed on that protocol but effectively has used it due to the saved computations (reason for wiping all data).

We could explore only wiping data dependent on protocols, but since login comes before loading data, it may be preferable to wipe all data.

ebrahimebrahim, 2025-04-24:
Agree that triggering a scene close (using the existing functions that do that) on change of user makes sense. But there should be a warning asking for confirmation, when someone tries to switch user while there are loaded openlifu objects around, so they don't unknowingly lose their work

Adding (3) here or later are both okay. I think it's in milestone 3 maybe

arhowe00, 2025-04-24:
> Agree that triggering a scene close (using the existing functions that do that) on change of user makes sense. But there should be a warning asking for confirmation, when someone tries to switch user while there are loaded openlifu objects around, so they don't unknowingly lose their work

Great! @sadhana-r @ebrahimebrahim , I will look myself to find those existing routines. But just because you guys may be more familiar, do you mind pointing me to any utilities you know of at the top of your head? Are there any comprehensive "unload all data" functions? (I was thinking of calling the same stuff as "close without saving" in the guided workflow)

ebrahimebrahim, 2025-04-24:
Yes, excatly the close without saving stuff -- it's [`clear_session`](https://github.com/OpenwaterHealth/SlicerOpenLIFU/blob/72afe5b235726b7d4f2ee7ec58f7c6469b37bbf3/OpenLIFUData/OpenLIFUData.py#L1377)

ebrahimebrahim, 2025-05-02:
Reopening because protocol permissions aren't being applied to session creation, which is where a user would select what protocol they are ultimately going to use:

[image]

PR #314 (2025-05-01): Add management and enforcement of protocol specific user permissions
Closes #262 

This PR adds:

- Adding user roles in the user management console (note: you can't set these when creating a new user; you just choose between operator and admin still)
- Allowing allowed roles to be configured on protocols
- Using the user roles of the currently logged in user to determine which protocols can be seen at any given time

This also implements the important considerations in #262, such as:
- Loaded data are removed when changing users

Note, the following item remains to be addressed at a future da [... cut]
PR #319 (2025-05-02): Prevent unallowed protocols being loaded through creating or loading sessions
Closes #262 

#262 was reopened due to some [incorrect behavior](https://github.com/OpenwaterHealth/SlicerOpenLIFU/issues/262#issuecomment-2847781778) around enforcement of allowed protocols given the user roles.

This PR prevents unallowed protocols from being shown in the session creator, and attempts to load sessions with unallowed protocols will fail. I tested this example by adding "operator" to the allowed roles of the example session and using the example restricted user and example operator to compare. Things [... cut]
closed 2025-05-02 by ebrahimebrahim