OpenwaterHealth/SlicerOpenLIFU #265: Don't clutter the log with LIFUInterface

ebrahimebrahim, 2025-04-15:
> once the device is connection and monitoring starts, the python console gets clogged with messages, preventing effective use. There should be a way to disconnect.

This comment applies to both test mode and non-test-mode.

When in test mode, we can just remove those print statements that currently just show us that LIFUInterface is being observed. Maybe we can comment them out for now and put a little code comment explaining that these would be useful to uncomment if the event loop stuff needs to be debugged.

When in non-test-mode, there are log messages coming from LIFUInterface from the openlifu layer. We may be able to suppress those (or reduce them) in SlicerOpenLIFU if we set the logging level appropriately for LIFUInterface's logger. If the clutter is in the root logger, however, then we should open an issue on OpenLIFU-python to adjust that.

arhowe00, 2025-04-23:
What level do you think we should expose? Here are some of the logging lines that I can see in LIFUInterface:
```sh
notebooks/LIFUTestWidget.py:            logger.warning("TX UART interface not found in LIFUInterface.")
notebooks/LIFUTestWidget.py:            logger.warning("HV UART interface not found in LIFUInterface.")
src/openlifu/io/LIFUInterface.py:logger = logging.getLogger(__name__)
src/openlifu/io/LIFUInterface.py:        logger.debug("Initializing TX Module of LIFUInterface with VID: %s, PID: %s, baudrate: %s, timeout: %s", vid, tx_pid, baudrate, timeout)
src/openlifu/io/LIFUInterface.py:        logger.debug("Initializing Console of LIFUInterface with VID: %s, PID: %s, baudrate: %s, timeout: %s", vid, con_pid, baudrate, timeout)
src/openlifu/io/LIFUInterface.py:            logger.error("Error starting monitoring: %s", e)
src/openlifu/io/LIFUInterface.py:            logger.error("Error stopping monitoring: %s", e)
src/openlifu/io/LIFUInterface.py:            logger.info("Loading %s...", solution_name)
src/openlifu/io/LIFUInterface.py:            logger.info("%s loaded successfully.", solution_name)
src/openlifu/io/LIFUInterface.py:            logger.error("ValueError: %s", v)
src/openlifu/io/LIFUInterface.py:            logger.error("Error loading %s: %s", solution_name, e)
src/openlifu/io/LIFUInterface.py:            logger.info("Turn ON HV")
src/openlifu/io/LIFUInterface.py:            logger.info("Start Sonication")
src/openlifu/io/LIFUInterface.py:                logger.info("Sonication started successfully.")
src/openlifu/io/LIFUInterface.py:                logger.error("Failed to start sonication.")
src/openlifu/io/LIFUInterface.py:            logger.error("ValueError: %s", v)
src/openlifu/io/LIFUInterface.py:            logger.error("Error Starting sonication: %s", e)
src/openlifu/io/LIFUInterface.py:            logger.info("Stop Sonication")
src/openlifu/io/LIFUInterface.py:                logger.info("Sonication stopped successfully.")
src/openlifu/io/LIFUInterface.py:                logger.error("Failed to stop sonication.")
src/openlifu/io/LIFUInterface.py:            logger.error("ValueError: %s", v)
src/openlifu/io/LIFUInterface.py:            logger.error("Error Stopping sonication: %s", e)

```

ebrahimebrahim, 2025-04-23:
We could do `logging.WARNING` for the level.
That should silence the info and debug messages, but still show us warnings and errors.
Then if things are still too noisy I'd say the issue is on the LIFUInterface side

PR #287 (2025-04-23): Dont clutter the log with lifu interface
Closes #265 

- [x] Since we can't test the HW, let's wait for @ebrahimebrahim to finish messing around with logging for meshroom stuff, so maybe after combining our PRs we assume everything will work out.

## For Review

Make sure the meshroom stuff is finished, then briefly inspect code. Note: Because there were not specifics in #265, I'm not sure whether I commented all the stuff out that was desired. Furthermore, is it enough just to set the logger for "LIFUInterface"?

You probably should not have to build and test anything.