OpenwaterHealth/SlicerOpenLIFU #267: Start doing integration testing

ebrahimebrahim, 2025-04-15:
We should generally start including integration testing in SlicerOpenLIFU.

Since we have zero of these at the moment, adding the first one is going to be the hardest part. So this issue represents just adding a single integration test of just anything, as long as its some multi-module workflow, to establish a precedent for future tests.

Some questions to be answered:
- Where will the test live? If it covers multiple modules, then is it logical for it to live [in OpenLIFUHome](https://github.com/OpenwaterHealth/SlicerOpenLIFU/blob/50a6804b582af8ade4118002c9c8ea31d4dd17c1/OpenLIFUHome/OpenLIFUHome.py#L262C7-L262C23) for example?
- What kind of example data will the test use? How will it get data if not from db_dvc?
- What is the best way to run the test? Is this something you should be able to run headless, or would that prevent it from messing around with widgets? What if there are many tests -- is there a convenient way to run all of them?

ebrahimebrahim, 2025-04-21:
I just remembered that I've had some snippets that I've used in the past to more conveniently test specific things I was working on or reviewing. They may be a helpful reference

- https://github.com/OpenwaterHealth/SlicerOpenLIFU/pull/159#pullrequestreview-2627813335
- https://github.com/OpenwaterHealth/SlicerOpenLIFU/pull/179#issuecomment-2676638535 (expand the "Details")

arhowe00, 2025-05-07:
Before I set us in a direction regarding our use of test data in SlicerOpenLIFU, we should weigh the options.  Is there a consensus desire to separate testing data for openlifu and SlicerOpenLIFU?  This works, but it may lead to hiccups such as version mismatches and forgetting to to double update data if we are not careful.

The current state of our data is that we have 'db_dvc' for human use, and openlifu uses git-controlled data in `tests/`, which is notably small and works seamlessly with pytest.

To adopt an automated workflow for testing SlicerOpenLIFU, we will need version controlled data, and tracking this in git *may* not work for us (e.g. if we use a large MRI). However an important question is whether we need our test data to scale. Does the size of the data impact the testing of the extension? If the answer is "no," it may save time and effort to track the data in git. It may be extra wise to reference the same data in `tests/` tracked by openlifu. This would technically be possible, since both SlicerOpenLIFU and openlifu use the same data schema, though some features are only intended for use in the extension.

If we must separate testing data, the consistency of testing data with the openlifu guaranteed by the openlifu hash will no longer apply. When updating the openlifu version, the SlicerOpenLIFU data will need to be updated (whether it is git-tracked or stored somewhere scalable). Now, we explore using scalable testing data stores:

## Using DVC with SlicerOpenLIFU

Similar to 'db_dvc', we would have version controlled test data for SlicerOpenLIFU. In OpenLIFULib, we would introduce utility functions that use the dvc API to collect this test data, with an API key supplied in an environment variable. When the individual modules are being tested, or just the OpenLIFUHome module (which calls the other tests), these utilities would be called. We could either hardcode a hash corresponding to specific versions of the dvc data, or we could make sure that when updating openlifu and the corresponding dvc test data, that those data updates don't break anything about SlicerOpenLIFU.  Therefore, updating the version of openlifu, which would come with a corresponding SlicerOpenLIFU test data update, might imply having to update SlicerOpenLIFU before doing a version bump.

## Potential workflow improvement: using DVC with pytest

If we decide to version control the testing data for the extension, one way we could improve our workflow is by referencing the same data in pytest for openlifu. Instead of using `tests/`, we would reference a unified data store whose test data is updated with openlifu updates and contains a comprehensive database. We can still include simple data for most pytest use cases, but we can also include larger data objects.

This would change how our tests are run in CI/CD pipelines. We could use a hook like the following:

```yaml
- name: Pull data with DVC
  run: dvc pull
- name: Run tests
  run: pytest tests/
```

which would be efficient in most cases because the data are cached. Then, for CI/CD, we could explore using GitHub Actions Secrets if we don't want the testing data to be public.  However, assuming we use a read-only public DVC remote, we could have something like:

```yaml
steps:
  - name: Set up DVC
    run: pip install 'dvc[gdrive]'  # or another remote

  - name: Pull data from DVC
    run: dvc pull
```

to fetch the data in the CI. We could then use similar tricks for automating data fetching in SlicerOpenLIFU, but let's consider that depending on what we choose.

---

Deciding on how we reference data depends on how much we care about data volume scalability. We could avoid using dvc altogether for everything testing related, eliminating the need for anything above, only if it is sufficient to test the extension.

ebrahimebrahim, 2025-05-26:
Unit tests and integration tests are doing different things, so there is a case for separating the test data here. It wouldn't be duplication, but quite different data that makes sense to use for integration tests.

Let's discuss more tomorrow to hash out some of the details, as well as whether we want to pursue this aggressively yet or not during this milestone.