OpenwaterHealth/SlicerOpenLIFU #282: Add setting parameter constraints and API for bounding values in protocol config

arhowe00, 2025-04-22:
After having merged the solution analysis PR in [OpenLIFU-python PR#250](https://github.com/OpenwaterHealth/OpenLIFU-python/pull/250), a clear definition of Parameter constraints was made. The Protocol Config module should create a custom form for this that allows definition of `ParameterConstraint`s. This definition should allow selection from operators and whether ranges or fixed values issue warnings or errors.

In addition, creating an API for fixing bounds of the dynamically created forms from [SlicerOpenLIFU PR#225](https://github.com/OpenwaterHealth/SlicerOpenLIFU/pull/225) will be helpful, as currently the bounds are chosen from a generic collection of magic numbers.

ebrahimebrahim, 2025-06-10:
@arhowe00 Is this complete? Close if so

closed 2025-06-10 by arhowe00

arhowe00, 2025-06-10:
Closed as part of #311 .