OpenwaterHealth/SlicerOpenLIFU #324: Validate protocols in the protocol config module

ebrahimebrahim, 2025-05-02:
In the protocol configuration module, a newly created protocol should run through a validation algorithm before the protocol can be published to the database via an instantiation validation in a Protocol's ` _post_init_` method, so that an error will be shown if an instantiated Protocol uses "out of bounds" attribute values.

arhowe00, 2025-05-05:
Hey, to clarify on this. Do you mean that `__post_init__` will add a validation method to `Protocol`, or that it *is* the validation method? Because I believe that this instantiation method is run as soon as the Protocol is configured and before publishing to the database.  If I were to add a validation, what are the validation constraints?

@peterhollender did send me a list of constraints which I set to prevent certain attributes from being configured, but this seems like it might be a more comprehensive validation.

arhowe00, 2025-05-06:
Okay, I have a better idea of what you want. Also to refresh on how this would work in the protocol configuration:

The protocol configuration allows configuration of all member objects referenced by `Protocol`, so things like `Pulse`, `Sequence`, `FocalPattern`, `ApodizationMethod`, etc. When the protocol is configured, each dataclass has *its own* `__post_init__` that is called and will handle all validation concerns that were made for those specific objects -- of which there are many across the Python codebase. 

`Protocol` itself does not have a `__post_init__`. It may not need one unless it needs to relate certain quantities to each other, such as the `FocalPattern` and `ApodizationMethod`. That is still called automatically after all the sub-object's `__post_init__` is called.

I cannot skip the `__post_init__` checks but I can catch any exceptions thrown and discard them unless the researcher wishes to publish a protocol to the database. That can be handled by nice exception handling and error messages. Is this a good approach?

ebrahimebrahim, 2025-05-06:
As discussed in our meeting just now, the goal is handle validation that takes place in any `__post_init__` that could be added later by catching any exceptions and showing the user reasonable errors. If the raising of exceptions upon construction of an invalid protocol causes problems for protocol editor caching, then this should be handled. One method you suggested was to subclass and override the `__post_init__` just for the caching.

peterhollender, 2025-05-07:
So one way that a protocol is instantiated is through loading from a
dictionary read from a json file. That `from_dict` method calls all of the
constructors for its various components. This relies on all of the init and
post_init validation, since the user could have typed literally anything
into the json file.

I’m not sure we need to instantiate the protocol or its attribute
dataclasses at all until the user has entered everything. Couldn’t it just
be building a dict, and then it instantiates via `from_dict` when you click
“save”? We could add a “check for errors” or something that would try to
instantiate it and tell you want the error is. Or is that misunderstanding
how the UI handles the data in the input fields?

ebrahimebrahim, 2025-05-07:
Yes that's another way to do it: switch to using a dict for caching. Currently it uses `Protocol` itself for caching. The UI doesn't inherently build a python dict or anything like that, Qt is lower level than that and really lives in the C++ layer-- you have to wire things up yourself. Andrew will see what's the simplest change to enable what's needed; in any case there are many options!

PR #332 (2025-05-14): Validate protocols in the protocol config module
Closes #324 

Newly created protocols run through validation algorithms defined in the `__post_init__`  method, both for the top-level `Protocol` class and all sub-objects. This PR officially handles thrown exceptions (i.e. for "out of bounds" attribute values) by displaying them to the researcher and failing to publish the Protocol to the database.

Caches skip the validation through an introduced utility `instantiate_without_post_init` that does not call `__init__`.

## For review

I already tested this by configuring protocols with values tha [... cut]
PR #353 (2025-05-20): Add protocol validity indicator
Closes #334 

This adds a simple validity indicator which checks that when a field is edited, whether it would produce an erroneous value. 

## For review:

A code review should be good enough, but you can also try dynamically changing some values in the protocol configuration and checking that any exceptions that would be thrown are reported at the bottom. You can try switching between protocols that have/don't have errors and fixing your errors, and *before saving*, you should know that your protocol has at least one error.

**Note**: We do not add the ability [... cut]
closed 2025-05-14 by ebrahimebrahim