OpenwaterHealth/SlicerOpenLIFU #341: Changing database location should log out user

arhowe00, 2025-05-12:
If an admin user changes the database path to another location and then explicitly reloads the database, the admin account should be logged out. This is to prevent somebody with admin privileges somewhere not being able to change the database to a location where that user doesn't know the credentials of an admin account. 

**Note: Doing this will mean that if an admin changes the database location to a place, they must be able to login as admin in that location to change it back.**

ebrahimebrahim, 2025-05-14:
> This is to prevent somebody with admin privileges somewhere not being able to change the database to a location where that user doesn't know the credentials of an admin account.

I think maybe you mixed up the phrasing here. Do you mean the following:

This is to prevent somebody with admin privileges somewhere _being_ able to change the database _path_ to a location _and gaining unauthorized admin access to the database at that location_

arhowe00, 2025-05-15:
> I think maybe you mixed up the phrasing here. Do you mean the following:
>
> This is to prevent somebody with admin privileges somewhere _being_ able to change the database _path_ to a location _and [...]

Yes!

closed 2025-05-20 by ebrahimebrahim