OpenwaterHealth/SlicerOpenLIFU #378: Don't warn about photocollection overwrites on session reloads

arhowe00, 2025-05-27:
Clicking loading the same session a few times will start to warn about replacing photocollection IDs. This warning is not needed.

arhowe00, 2025-06-12:
Issue closed due to incompatible preferences and changes to the data module

arhowe00, 2025-06-16:
To reproduce:
```python
slicer.modules.OpenLIFUDatabaseWidget.onLoadDatabaseClicked(True)
slicer.modules.OpenLIFUDataWidget.logic.subject = slicer.modules.OpenLIFUDatabaseWidget.logic.db.load_subject('soren')
def load_soren_session():
  slicer.modules.OpenLIFUDataWidget.logic.load_session(
    subject_id='soren',
    session_id='test session',
  )
load_soren_session()
load_soren_session()
slicer.modules.OpenLIFUDataWidget.logic.clear_session()
```

ebrahimebrahim, 2025-06-17:
> I think the reason for the checks is because the clear_session() doesn't clear the scene?

Clear session is meant to clear the session related data only, because SlicerOpenLIFU allows you to load in transducers and protocols manually without a session and we don't want to destroy that data on a session unload.

However there is no way to load a photocollection _manually_.

And even if there were, we never really _load_ them... we are just creating a list of reference numbers.

So we don't need to be checking about whether a photocollection is about to get "replaced".

Not sure if I understood your question or answered it, lmk if not