OpenwaterHealth/SlicerOpenLIFU #430: Peak Negative Pressure replaces Photoscan in Preview and Tracking

peterhollender, 2025-06-18:
When you have "Render Peak Negative Pressure" checked, and generate a new photoscan, the "Preview Photoscan" button pulls up this: 

[image]

Also shows up in the transducer tracking wizard, rendering it unusable.

If I hide the peak negative pressure, the photoscan comes back and works fine.

sadhana-r, 2025-06-27:
Actually, I played around with it a bit more today going back and forth between the different modules and I was able to reproduce the issue! It happens when you preview a photoscan, compute a sonication solution, render the pnp and then go back and view the same photoscan. I have figured out which part of the code is causing the issue so will push a fix soon.

peterhollender, 2025-06-27:
Whew, thanks @sadgana-r!

PR #439 (2025-06-27): Hide pnp rendering from photoscan preview and tracking wizard
Closes #430 

The issue occured when a photoscan was previewed before a sonication solution was computed and rendered. Subsequent previews of the same photoscan incorrectly displayed the PNP rendering. 
By default, the pnp rendering display node includes the view node associated with the preview/tracking window.  This PR updates the way display nodes are hidden from certain views, by explicitly accounting for the case where a display node is associated with a non-empty list of view node IDs. 

For review:
- Confirm that the issue is [... cut]
closed 2025-06-27 by ebrahimebrahim