OpenwaterHealth/SlicerOpenLIFU #433: PNP/Intensity Color Controls

peterhollender, 2025-06-25:
When rendering Peak Negative Pressure, it would be good to have some controls for setting the color and opacity bounds. These can already be controlled through the `Volumes` and `Volume Rendering` modules now, but some of those basic controls should be available on the Sonication Planning page. 

## Color 
- Double Handle (Manual Min/Max) Slider
- Slider Minimum = 0
- Slider Maximum = Target Pressure
- Default to (0, Target Pressure)

## Opacity
- Double Handle (Manual Min/Max) Slider
- Slider Minimum = 0
- Slider Maximum = Target Pressure
- Default to (0.1*Target Pressure, 0.5*Target Pressure)
- Bounds set the 0 and 100% opacity limits in volume rendering, as well as the lower bound for thresholding of slices

peterhollender, 2025-07-16:
@ebrahimebrahim Could we get someone to work on this? I think its very confusing for users when automatic thresholding clips the top end of the color scale to something much less than the peak pressure, resulting in a significant over-representation of the sidelobes.

arhowe00, 2025-08-09:
Hey Peter, I’m a bit unclear on the exact behavior you want here. After computing a solution, in "Volume" and "VolumeRendering," I see two volumes — `intensity` and `p_min` — and the only control that seems to visibly affect things is the "Shift" for `p_min`.

Could you clarify which volume(s) these controls should target, and what parameters they should adjust to achieve the effect you want?

ebrahimebrahim, 2025-08-11:
Hi Andrew, I think I can clarify here:

- We only show p_min, so these controls are all about p_min
- The parameters these adjust are the color mapping and the opacity mapping. I'll help you look into adjusting those programmatically. The shift slider you are talking about gets at one aspect of this, but there is more. Hit me up and we can chat about this.

peterhollender, 2025-08-11:
Thanks, @ebrahimebrahim, that's exactly right. Currently we see this: 

[image]

And if I turn on the overlay in the slices(which should probably come on when checking the box?), we can see that it's very saturated:

[image]

We already know what the expected value range is from the protocol (and the max value of the p_min array), but the default Display settings for `p_min` in the `Volumes` module are terrible:

[image]

 Simply setting them both to (0, target_pressure), and then checking "Synchronize with Volumes Module" in Volume Rendering) is a huge help:

[image]

[image]

I might revise my request on this a bit to make it easier. Maybe we can have a double sided slider that controls the "Manual Min/Max" for "Window/Level" on the color controls, and a single-sided slider for the lower Manual Threshold on the Opacity. I think that having gone to maximum intensity projection, we don't need to scalar opacity control in the volume rendering.

PR #475 (2025-08-25): Add peak negative pressure window level / opacity threshold controls
Closes #433 

The automatic rendering of the Peak Negative Pressure (p_min) volume can obscure the true pressure field distribution due to suboptimal defaults.

This change allows to manually defining the window/level for the color and opacity. This allows for a more precise and accurate visualization of the underlying simulation data.
closed 2025-08-25 by ebrahimebrahim