OpenwaterHealth/SlicerOpenLIFU #458: Images Location

stahir715, 2025-07-18:
The images on the old version of the photogrammetry application we're using were located in the DCIM folder, however they've moved to the "Internal Storage" folder. We should be able to locate these images in the new folder (or potentially any other folder they get moved to?) and discover where they are. (@arhowe00 for some reason I can't assign you in the right column so I'm just tagging you here)

arhowe00, 2025-08-10:
We can make the import path configurable, but turning this into a full file explorer could be challenging. Accessing the Android file system already requires a bridging implementation between Android and the host computer (macOS, Windows, Linux). There already is friction with needing to set up the android debugger, since the 3D Open Water app stores the photos on disk and expects SlicerOpenLIFU, the receiver, to configure that path and grab the photos.

I can make a simple field that allows you to type in an alternative android path -- which does depend on the particular implementation serving the android filesystem, but is the same cross platform when using Google's adb. Exposing this field to the user seems quite a bit low level though, I don't think it would be difficult to hardcode the path change when things change.

I'm not sure yet whether we can have an "Open the android file system" button to search the device yet. It could require more dependencies. I really think a cloud storage instance is the way here.

@peterhollender let us know if I should a field where you can type in the android path

peterhollender, 2025-08-10:
I don’t need navigation, but I think the simple text entry of the relative
android path that adb uses to find the location is acceptable in the slicer
module, at least for now. It can be hidden in the custom app